lit-robotics/libcamera-rs #1: Video Recording / Streaming

Maufeat, 2023-01-15:
First off, thanks for creating this binding.

Is there any example on how I can start recording? I use the ArduCam 64MP Hawk-Eye. MJPEG is natively not supported. Just NV12, RGB888, XRGB8888

Thanks for any help :)

chemicstry, 2023-01-16:
Hey,

The [jpeg_capture example](https://github.com/lit-robotics/libcamera-rs/blob/main/libcamera-rs/examples/jpeg_capture.rs) forces MJPEG format so that frame data could be saved directly into `.jpeg` file without any encoding. If you removed [this assert](https://github.com/lit-robotics/libcamera-rs/blob/main/libcamera-rs/examples/jpeg_capture.rs#L48), camera would probably generate a configuration with one of the supported formats. It is up to you, however, to save the raw data in an appropriate format that can be used somewhere else. `libcamera` is quite a "low level" library after all. If you looked at [libcamera-apps](https://github.com/raspberrypi/libcamera-apps) they have a lot of code dedicated to encoding.

Regarding streaming/recording, you just requeue capture requests when they are completed to produce a continuous stream of frames.

chemicstry, 2023-02-15:
There is now a recording example with #5

closed 2023-02-15 by chemicstry