lit-robotics/libcamera-rs #21: Video capture yuv420 to rgb

thedodd, 2023-05-07:
Hey folks, stoked to see this project here. I have a use case for which I would like to start using this project, instead of directly using libcamera-raw/libcamera-vid.

In essence, I would like to capture video off of my picamera 2, and convert that data to rgb format. Out of the box, libcamera-vid supports few formats and yuv420 seems to be the base format coming out of the ISP.

Do folks have an recommended way of handling this using this library? Perhaps the mjpeg video example which is already in the repo is the best place to start, and:
- instead of mjpeg, use yuv420,
- when a frame is received, manually convert from yuv420 to rgb,
- emit convert rgb output from there.

Thoughts?

chemicstry, 2023-05-08:
Hi,

If I remember correctly, raspberry ISP should support RGB output. You can use `list_cameras` example from this repository and it will print all supported formats. I think the official libcamera-apps also have an option to print supported formats.

If RGB is supported, then you can just set the format to RGB instead of MJPEG in video example and it should work. Although I haven't done much testing on formats with multiple data planes (MJPEG has just one, while RGB has 3), so report if you encounter any issues.

If RGB is not supported then yes, I think you can just capture YUV420 and convert it yourself. This will be slower than with ISP, but it should still be much faster than encoding/compressing.

frederikstroem, 2023-11-07:
> Hey folks, stoked to see this project here. I have a use case for which I would like to start using this project, instead of directly using libcamera-raw/libcamera-vid.
>
> In essence, I would like to [...]

I am using the Raspberry Pi Camera Module 2 (8MP) [IMX219] on a Raspberry Pi 4 with 8GB RAM. I was unable to get MJPEG working with the camera and found RGB888 to be the best alternative for my use case. However, I did find the following formats available for this camera:
- NV21
- YUV420
- NV12
- XRGB8888
- BGR888
- RGB888
- XRGB8888
- RGB565
- YVYU
- YUYV
- VYUY
- UYVY

To get the camera working with RGB888, I modified the [`jpeg_capture.rs`](https://github.com/lit-robotics/libcamera-rs/blob/56e2442986bdc4d9be89193bb72254ae8cf969ff/libcamera/examples/jpeg_capture.rs) example to utilize RGB888 instead of MJPEG.

```rust
use drm_fourcc::DrmFourcc;

use image::RgbImage;
use image::ColorType;
use image::ImageBuffer;

use std::time::Duration;

use libcamera::{
    camera::CameraConfigurationStatus,
    camera_manager::CameraManager,
    framebuffer::AsFrameBuffer,
    framebuffer_allocator::{FrameBuffer, FrameBufferAllocator},
    framebuffer_map::MemoryMappedFrameBuffer,
    pixel_format::PixelFormat,
    properties,
    stream::StreamRole,
    geometry::Size,
};

const PIXEL_FORMAT_RGB888: PixelFormat = PixelFormat::new(DrmFourcc::Rgb888 as u32, 0);

fn main() {
    let filename = "rgb888_capture.png";

    // Set size.
    let size = Size { width: 800, height: 600 };

    // Initialize the camera manager and select the first camera.
    let mgr = CameraManager::new().unwrap();
    let cameras = mgr.cameras();
    let cam = cameras.get(0).expect("No cameras found");
    println!(
        "Using camera: {}",
        *cam.properties().get::<properties::Model>().unwrap()
    );

    // Acquire the camera.
    let mut cam = cam.acquire().expect("Unable to acquire camera");

    // This will generate default configuration for each specified role.
    let mut cfgs = cam.generate_configuration(&[StreamRole::ViewFinder]).unwrap();

    // Set the desired pixel format and size for the configuration.
    cfgs.get_mut(0).unwrap().set_pixel_format(PIXEL_FORMAT_RGB888);
    cfgs.get_mut(0).unwrap().set_size(size);

    // Print the generated configuration.
    println!("Generated config: {:#?}", cfgs);
[... 62 lines cut ...]
    image::save_buffer(&filename, &img, size.width, size.height, ColorType::Rgb8).unwrap();

    println!("PNG file saved to {}", &filename);

    // Everything is cleaned up automatically by Drop implementations.
}
```

Make sure to include drm-fourcc, image, and, lastly, libcamera manually via Git (for the Raspberry Pi, see issue #33) in the Cargo.toml file.

```toml
[package]
name = "app"
version = "0.1.0"
edition = "2021"

[dependencies]
drm-fourcc = "2.2"
image = "0.24.7"
libcamera = { git = "https://github.com/lit-robotics/libcamera-rs.git" }
```

What threw me off at first was that the RGB data is interleaved as a single plane, rather than being split into three separate planes for each color channel, as @chemicstry hinted. For more insight, see [here](http://avitevet.com/uncategorized/when-to-use-it-interleaved-vs-planar-image-data-storage/) ([archive](https://web.archive.org/web/20231107183033/http://avitevet.com/uncategorized/when-to-use-it-interleaved-vs-planar-image-data-storage/)).

**P.S.** @thedodd, I'm aware that you were referring to video, but I haven't reached that part of my project yet. Nevertheless, I thought sharing my findings might be helpful, as they could also apply to the video implementation.

alexisgaziello, 2024-02-29:
Hi, I am using the camera module2 and raspberry pi and converting the NV12 format via [dcv-color-primitives](https://docs.rs/dcv-color-primitives/latest/dcv_color_primitives/). NV12 is a YUV format so this might help.

Code example:

```
use dcv_color_primitives as dcp;
use dcp::{convert_image, ColorSpace, ImageFormat, PixelFormat};
use image::{ColorType, ImageEncoder};

pub(crate) fn nv12_to_rgb(planes: &Vec<&[u8]>, width: u32, height: u32) -> Result<Vec<u8>, &'static str> {
    let src_format = ImageFormat {
        pixel_format: PixelFormat::Nv12,
        color_space: ColorSpace::Bt601,
        num_planes: 2, // NV12 has two planes: Y (luminance) and UV (chrominance)
    };

    let dst_format = ImageFormat {
        pixel_format: PixelFormat::Rgb,
        color_space: ColorSpace::Rgb,
        num_planes: 1,
    };

    let mut dst_data = vec![0u8; (3 * width * height) as usize]; // RGB output

    convert_image(
        width,
        height,
        &src_format,
        None,
        &[planes[0], planes[1]], // Y and UV planes
        &dst_format,
        None,
        &mut [&mut dst_data],
    ).map_err(|_| "Failed to convert NV12 to RGB")?;

    Ok(dst_data)
}

```

Which you can use by just passing the buffer data, taking the example from the [repo](https://github.com/lit-robotics/libcamera-rs/blob/df2d09f9c03aa5a74352197e076405c1010a6cff/libcamera/examples/jpeg_capture.rs#L102):

```
    // For NV12, we expect two data planes: Y (luminance) and UV (chrominance)
    let planes = framebuffer.data();

    // Convert NV12 to RGB
    let rgb_data = nv12_to_rgb(&planes, width, height).unwrap();
```

Its my first time writing rust so take this code carefully.

chemicstry, 2025-05-26:
Closing as it is not an issue with libcamera-rs

closed 2025-05-26 by chemicstry