lit-robotics/libcamera-rs #22: Setting Camera Controls

Eldir1, 2023-05-10:
Hello,

thank's a lot for your great work.
Is setting of camera controls currently not supported or did I use it the wrong way?

```
let mut ctrl_list = ControlList::new();
ctrl_list.set(libcamera::controls::AwbEnable(false)).expect("Error setting AwbEnable");
// ... also tried some other controls
cam.start(Some(ctrl_list.deref()));
```
The original documentation of libcamera explains, that ControlList should be constructed from ControlInfoMap, but this class is currently in the Rust part only poorly supported. So I couldn't use ControlInfoMap to read the content of the map.

:smile:

Sc0rpe, 2023-05-12:
I'd also like to do this. I'v tried to set exposure time as this:
`           let mut ctrl_lists = ControlList::new();
            let exp_time: SmallVec<[i32; 1]> = smallvec![100000];           // 100 ms
            ctrl_lists.set::<ExposureTime>(ExposureTime::try_from(ControlValue::Int32(exp_time)).unwrap());            
            actcam.start(Some(&ctrl_lists)).unwrap();            `

However, when running like this i get Errors from IPCPipe after allocating the buffers. 
`[3:06:31.495482900] [5654] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
[3:06:31.495603509] [5654] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[3:06:31.495656138] [5654] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:253 Failed to call start
[3:06:31.503491688] [5654] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Connection refused
[3:06:31.503593556] [5654] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[3:06:31.503647481] [5654] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:438 Failed to call mapBuffers
`
I get the same errors even when i didn't set anything, just passing the controlList to start() method.

kbingham, 2023-05-12:
> However, when running like this i get Errors from IPCPipe after allocating the buffers. `[3:06:31.495482900] [5654] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout! [3:06:31.495603509] [5654] [...]

This looks like a distinct issue that you have IPA modules isolated which aren't working.

chemicstry, 2023-05-13:
Please check issue #2 and specifically this comment https://github.com/lit-robotics/libcamera-rs/issues/2#issuecomment-1430644226

Are you sure that your camera supports the specified control? Currently it will just silently fail. There's definitely some improvements to be done.

Sc0rpe, 2023-05-15:
I am using a Raspberry Pi 4 with the new Raspberry Pi Global Shutter Camera ( IMX296 ).
I tried with following control settings: ExposureTime, FrameDuration, MeteringMode, DigitalGain, AnalogueGain.
None of them worked for me. I get the same error i described befor for all of them.
Even when i don't add any setting to the control list, and simply pass the empty list to the start() function i get these errors.

chemicstry, 2023-05-15:
@Sc0rpe sorry I was referring to @Eldir1's issue regarding controls not working. Yours seems to be a different issue as @kbingham described. Does it work with `start(None)`? If not, then it is not related to controls, you have issues with libcamera installation.

kbingham, 2023-05-15:
If you can share a sample app/code that crashes/fails it would help for replicating the issue too.

kbingham, 2023-05-17:
Indeed I don't expect this to be specific to the camera, it's how you've installed or set up libcamera (or perhaps how someone else like the distro has).

Are you using libcamera from raspbian? Or did you build it yourself?

Sc0rpe, 2023-05-17:
I am running raspbian bullseye 32-bit lite.
As the documentation on https://crates.io/crates/libcamera stated, i installed libcamera v0.0.4 myself.

kbingham, 2023-05-17:
Ok - I would suspect that the reason you have IPA failures is because you didn't have all the required dependencies installed when you built and installed libcamera v0.0.4 yourself.

Sc0rpe, 2023-05-17:
@kbingham Thanks! You were right.
I ran `meson --reconfigure build` in my libcamera folder source directory.
And i saw a message that `gnutls` and `libcrypto` where missing. Thats why the IPA modules ran in isolated mode.
I installed the missing dependencies and built libcamera again. Now everything works fine.
Thank you very much! :)

chemicstry, 2023-06-05:
@Eldir1 did you solve your issue?

Eldir1, 2023-06-19:
Thank's for this good support!

closed 2023-06-19 by Eldir1