lit-robotics/libcamera-rs #25: Fails to compile in a docker image

UnknownError55, 2023-07-06:
This is the dockerfile I am using to try to compile libcamera-rs. With Cargo.toml having version 0.2.1 of libcamera-rs. The architecture I am building on in aarch64.
```Dockerfile
# syntax=docker/dockerfile:1.5-labs
FROM rust:bullseye as builder

# ===== Install all needed packages
RUN apt update -qq && apt -y install \
  build-essential \
  pkgconf \
  python3 \
  python3-pip \
  ninja-build \
  meson \
  git \
  cmake

RUN pip3 install --user meson jinja2 ply jinja2 pyyaml && \
  pip3 install --user --upgrade meson

ADD --keep-git-dir=true https://git.libcamera.org/libcamera/libcamera.git /libcamera/
RUN cd /libcamera/ && \
  meson setup build && \
  ninja -C build install

# ===== Attempt to install all the packaged we need
WORKDIR /app
COPY Cargo.toml .
RUN mkdir src && echo "fn main() {}" > src/main.rs && cargo build
```

The error trying to compile the docker image. Looks like there might be an issue with the plurality of `StreamRole` / `StreamRoles`? And an error with the scoping of `roles_vec`? Which is rather odd since the original cpp library should compile (and does with the `ninja -C build install` step). So I am not sure what grievances it has with the cpp source code.
```
#0 41.47 The following warnings were emitted during compilation:
#0 41.47
#0 41.47 warning: c_api/camera.cpp: In function 'libcamera_camera_configuration_t* libcamera_camera_generate_configuration(libcamera_camera_t*, const libcamera_stream_role*, size_t)':
#0 41.47 warning: c_api/camera.cpp:70:16: error: 'StreamRoles' is not a member of 'libcamera'; did you mean 'StreamRole'?
#0 41.47 warning:    70 |     libcamera::StreamRoles roles_vec((libcamera::StreamRole*)roles, (libcamera::StreamRole*)roles + role_count);
#0 41.47 warning:       |                ^~~~~~~~~~~
#0 41.47 warning:       |                StreamRole
#0 41.47 warning: c_api/camera.cpp:71:46: error: 'roles_vec' was not declared in this scope
#0 41.47 warning:    71 |     return cam->get()->generateConfiguration(roles_vec).release();
#0 41.47 warning:       |                                              ^~~~~~~~~
#0 41.47 warning: c_api/camera.cpp:69:134: warning: unused parameter 'roles' [-Wunused-parameter]
#0 41.47 warning:    69 | libcamera_camera_configuration_t *libcamera_camera_generate_configuration(libcamera_camera_t *cam, const enum libcamera_stream_role *roles, size_t role_count) {
#0 41.47 warning:       |                                                                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
#0 41.47 warning: c_api/camera.cpp:69:148: warning: unused parameter 'role_count' [-Wunused-parameter]
#0 41.47 warning:    69 | libcamera_camera_configuration_t *libcamera_camera_generate_configuration(libcamera_camera_t *cam, const enum libcamera_stream_role *roles, size_t role_count) {
#0 41.48 warning:       |                                                                                                                                             ~~~~~~~^~~~~~~~~~
#0 41.48
#0 41.48 error: failed to run custom build command for `libcamera-sys v0.2.1`
#0 41.48
#0 41.48 Caused by:
#0 41.48   process didn't exit successfully: `/app/target/debug/build/libcamera-sys-79d71b9dfcb27915/build-script-build` (exit status: 1)
#0 41.48   --- stdout
#0 41.48   cargo:rerun-if-env-changed=LIBCAMERA_NO_PKG_CONFIG
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_aarch64-unknown-linux-gnu
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_aarch64_unknown_linux_gnu
#0 41.48   cargo:rerun-if-env-changed=HOST_PKG_CONFIG
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG
#0 41.48   cargo:rerun-if-env-changed=LIBCAMERA_STATIC
#0 41.48   cargo:rerun-if-env-changed=LIBCAMERA_DYNAMIC
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_aarch64-unknown-linux-gnu
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_aarch64_unknown_linux_gnu
#0 41.48   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_PATH
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_aarch64-unknown-linux-gnu
#0 41.48   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_aarch64_unknown_linux_gnu
#0 41.48   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
[... 78 lines cut ...]
#0 41.48   running: "c++" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-I" "/usr/local/include/libcamera" "-Wall" "-Wextra" "-std=c++17" "-o" "/app/target/debug/build/libcamera-sys-2ee62eac911404e4/out/c_api/camera.o" "-c" "c_api/camera.cpp"
#0 41.48   cargo:warning=c_api/camera.cpp: In function 'libcamera_camera_configuration_t* libcamera_camera_generate_configuration(libcamera_camera_t*, const libcamera_stream_role*, size_t)':
#0 41.48   cargo:warning=c_api/camera.cpp:70:16: error: 'StreamRoles' is not a member of 'libcamera'; did you mean 'StreamRole'?
#0 41.48   cargo:warning=   70 |     libcamera::StreamRoles roles_vec((libcamera::StreamRole*)roles, (libcamera::StreamRole*)roles + role_count);
#0 41.48   cargo:warning=      |                ^~~~~~~~~~~
#0 41.48   cargo:warning=      |                StreamRole
#0 41.48   cargo:warning=c_api/camera.cpp:71:46: error: 'roles_vec' was not declared in this scope
#0 41.48   cargo:warning=   71 |     return cam->get()->generateConfiguration(roles_vec).release();
#0 41.48   cargo:warning=      |                                              ^~~~~~~~~
#0 41.48   cargo:warning=c_api/camera.cpp:69:134: warning: unused parameter 'roles' [-Wunused-parameter]
#0 41.48   cargo:warning=   69 | libcamera_camera_configuration_t *libcamera_camera_generate_configuration(libcamera_camera_t *cam, const enum libcamera_stream_role *roles, size_t role_count) {
#0 41.48   cargo:warning=      |                                                                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
#0 41.48   cargo:warning=c_api/camera.cpp:69:148: warning: unused parameter 'role_count' [-Wunused-parameter]
#0 41.48   cargo:warning=   69 | libcamera_camera_configuration_t *libcamera_camera_generate_configuration(libcamera_camera_t *cam, const enum libcamera_stream_role *roles, size_t role_count) {
#0 41.48   cargo:warning=      |                                                                                                                                             ~~~~~~~^~~~~~~~~~
#0 41.48   exit status: 1
#0 41.48
#0 41.48   --- stderr
#0 41.48
#0 41.48
#0 41.48   error occurred: Command "c++" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-I" "/usr/local/include/libcamera" "-Wall" "-Wextra" "-std=c++17" "-o" "/app/target/debug/build/libcamera-sys-2ee62eac911404e4/out/c_api/camera.o" "-c" "c_api/camera.cpp" with args "c++" did not execute successfully (status code exit status: 1).
#0 41.48
#0 41.48
#0 41.48 warning: build failed, waiting for other jobs to finish...
```

chemicstry, 2023-07-08:
I believe this is due to recent change in libcamera: https://github.com/kbingham/libcamera/commit/26a4b83d1b9443795dbcb21fa6c23117f113216a

I will fix this, but I need to figure out what strategy to use to be in sync with libcamera versions.

In the meantime you could use libcamera version prior that commit, or try fixing rust bindings, PRs welcome!

chemicstry, 2023-08-03:
This should be fixed with #27

closed 2023-08-03 by chemicstry