lit-robotics/libcamera-rs #36: unable to set AwbEnable control

smroid, 2024-06-08:
I am able to set request controls such as AeEnable, ExposureTime, AnalogueGain. However, when I try to set AwbEnable (true or false), I get:

../include/libcamera/controls.h:152: T libcamera::ControlValue::get() const [with T = float; typename std::enable\
_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remov\
e_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> = nullptr]: Assertion `type_ == det\
ails::control_type<std::remove_cv_t<T>>::value' failed.

Any suggestions?

aaronyeeio, 2024-10-11:
I have the same issue here. The case for me is that I can't set `FrameDurationLimits` and it throws the same error as above.

aaronyeeio, 2024-10-11:
I wonder if this is caused by that `libcamera` has updated to v0.3.0. But I checked the official API compatibility across v0.1.0 to v0.3.0 and it shows a 97% coverage. And `include/libcamera/controls.h` also remains the same on the line it throws.

chemicstry, 2024-10-14:
This is duplicate of #31. I will check what I can do about that issue.

closed 2024-10-14 by chemicstry