lit-robotics/libcamera-rs #37: Set rotation

fishrockz, 2024-08-07:
How do i set the rotation?

From your examples i can see the rotation

```rust
    println!("Properties: {:#?}", cam.properties());
```

But this is a read only.

I had a go at trying to use the configuration but i could not figure it out.

chemicstry, 2024-08-13:
Rotation is a readonly property, like serial number. For example set by the laptop manufacturer if sensor is mounted upside down.

You can use opencv or similar image library to rotate image after capturing.

closed 2024-08-19 by chemicstry

perara, 2024-11-13:
There is a property orientation which is accessible in C++. This allows to tune orientation: Sample code:

`switch (cameraConf.camera_rotate) {
            case 0:

                cameraConfig->orientation = libcamera::Orientation::Rotate0;
                break;
            case 90:
                cameraConfig->orientation = libcamera::Orientation::Rotate90;
                break;
            case 180:
                cameraConfig->orientation = libcamera::Orientation::Rotate180;
                break;
            case 270:
                cameraConfig->orientation = libcamera::Orientation::Rotate270;
                break;
            default:
                SPDLOG_WARN("Invalid camera rotation value: {}. Defaulting to Rotate0.", cameraConf.camera_rotate);
                cameraConfig->orientation = libcamera::Orientation::Rotate0;
                break;
        }
        libcamera::StreamConfiguration& streamConfig = cameraConfig->at(0);`

 I could not find this exposed?

fishrockz, 2024-11-17:
I spoke to some libcamera people and they said similar.

Its actually fairly simple todo your self in software, in the end i just swapped the orientation of my sensor.. lol

Apparently the option in @perara 's code is not always changeable depending on the hardware so braking it out is of limited value but i do believe it should be possible..

however for all of the reasons above i didn't bother looking into this any further.