lit-robotics/libcamera-rs #39: Cannot build arm64 (aarch64) since the new 2.3.0 version

ItzSiL3Nce, 2024-10-29:
Before upgrading to version 0.3.0, I could build an application using libcamera-rs as a dependency for aarch64.
With the new version I get linking errors in libcamera saying that they expect a elf64-x86-64 version of libcamera.
For now, I'm sticking to version 0.2.3 which works perfectly, but it's missing the updated constants.

chemicstry, 2024-11-05:
Could you give more information about your build setup? Are you cross-compiling on a x86 host?

fishrockz, 2024-12-04:
So when cross compiling we have to be very careful what we build depend on, as it must be build able on the host.

when i cross compile i typically only have libcamera for the target so i can not build main line of this repo..

this repo depends on libcamera-sys for both dev (running build.sh) and build (building the final bin/lib)

I was able to cheat and get round this with https://github.com/lit-robotics/libcamera-rs/compare/main...fishrockz:libcamera-rs:willsalmon/extra-include but its  a bit hacky, i hope to find a nicer way todo this and submit a patch.

closed 2025-05-26 by chemicstry