lit-robotics/libcamera-rs #40: DMA for RPI.

perara, 2024-11-03:
Great library! 
However I could not identify any good ways of using DMA to exploit hardware encoding, such as H264/H265.

Is there any plans for supporting DMA?

chemicstry, 2024-11-03:
Hi, this library is just a memory safe interface to libcamera and any data processing is outside its scope. You could use something like [v4l](https://crates.io/crates/v4l) to encode raw data from libcamera. This is what official rpicam apps use (except in C++).

perara, 2024-11-05:
Dear @chemicstry 

I fully agree and I apologize if I was unclear.

What I'm referring to is the [DMAHeap ](https://libcamera.org/api-html/classlibcamera_1_1DmaHeap.html) that is part of the libcamera stack. This was released earlier this year I believe: https://patchwork.libcamera.org/patch/20102/

Here is an example of how it is used:

```cpp 
   std::map<libcamera::Stream *, std::vector<std::unique_ptr<libcamera::FrameBuffer>>> frame_buffers;

void createFramebuffer() {
        allocator = std::make_unique<libcamera::FrameBufferAllocator>(camera);

        for (libcamera::StreamConfiguration &config: *cameraConfig) {

            libcamera::Stream *stream = config.stream();
            std::vector<std::unique_ptr<libcamera::FrameBuffer>> fb;

            for (unsigned int i = 0; i < config.bufferCount; i++) {
                std::string name("capture-fb-" + std::to_string(i));
                libcamera::UniqueFD fd = dma_heap.alloc(name.c_str(), config.frameSize);

                if (!fd.isValid()) {
                    throw std::runtime_error("failed to allocate capture buffers for stream");
                }

                std::vector<libcamera::FrameBuffer::Plane> plane(1);
                plane[0].fd = libcamera::SharedFD(std::move(fd));
                plane[0].offset = 0;
                plane[0].length = config.frameSize;

                fb.push_back(std::make_unique<libcamera::FrameBuffer>(plane));

                void *memory = mmap(
                    nullptr, 
                    config.frameSize, 
                    PROT_READ | PROT_WRITE, MAP_SHARED, 
                    plane[0].fd.get(),
                    0
                    );

                mapped_buffers[fb.back().get()]
                        .emplace_back(static_cast<uint8_t *>(memory), config.frameSize);

            }

            frame_buffers[stream] = std::move(fb);

        }

    }```

chemicstry, 2024-11-05:
Oh, sorry for misinterpreting :D

I didn't know about DMAHeap in libcamera. If I understand correctly, this can be used to directly transfer data from libcamera to i.e. h264 encoder without copying buffers in rust application, right? Neat.

This is definitely something I would like to support, however, the project I was using libcamera for is on hold and I'm not sure when I will get to implementing something like this. But PRs are welcome and I would review it and merge it.

kbingham, 2024-11-05:
I'm not sure what libcamera-rs is doing to handle buffer management but libcamera only uses dmabuf handles for exactly the purpose of being able to zero-copy share the buffers to an encoder or display subsystem.

the DMAHeap's referenced above are /internal/ in libcamera and not exposed on the libcamera API - so I wouldn't expect the heap allocated to be part of a libcamera-rs API specifically either.

That said - it could be something a rust application uses to allocate memory to give to libcamera with externally allocated buffers that it will then use to encode. But there's a lot of possibilities here and I don't know what you're making to provide concrete advice.

But essentially - if you ask libcamera to allocate buffers for you - then you're already getting a dma buffer.

perara, 2024-11-13:
I've now investigated this and the way it ended was to retrieve the fd as you said and using mmap / munmap. This way i were able to send directly the v4l2m2m encoder on the RPI! 

But your were correct regarding DMAHeap!

closed 2024-11-13 by perara