lit-robotics/libcamera-rs #42: Multi pipeline example

fishrockz, 2024-11-17:
I have ended up setting up muplitple processing pipelines for pi cameras a few time, IIRC the python libs mention it and thats were i started

This is nice for saving a video and getting pictures for a view finder or tracking at the same time.

I have been looking into something similar with libcamera-rs

I think i just need to add a second value to the array passed to generate_configuration
https://gitlab.com/pointswaves/pi-camera/-/blob/main/camera-app/src/main.rs?ref_type=heads#L75

But i have not got this to work so far. My issues have been finding combinations of configs that with validate https://gitlab.com/pointswaves/pi-camera/-/blob/main/camera-app/src/main.rs?ref_type=heads#L95

Does anyone have any examples of getting this to work?

I would like to get the preview feed to export in as close to the native res of my view finder as possible, while also having a second pipeline at as high a quality as possible.

I have found interacting with pi cameras in rust much nicer than any of my previous projects in other languages, so i would like to say a big thanks to everyone involved in this lib!

But I have not found many examples of people using this lib to learn from so any examples people have with multiple pipelines would be greatly appreciated.

fishrockz, 2024-12-04:
I now have a example for this https://gitlab.com/pointswaves/pi-camera/-/blob/trying/camera-app/crates/pi-camera/src/main.rs?ref_type=heads I'm working on getting it into main of my repo https://gitlab.com/pointswaves/pi-camera

Would it be helpful to get a basic version of this into the examples? or shall i just keep making my repo better to the point were maybe this repo might point to it as a example?

chemicstry, 2024-12-05:
Thanks for figuring this out, a simple example of how to run multiple pipelines would be appreciated!