lit-robotics/libcamera-rs #43: Is there a reason for not having a lock file?

fishrockz, 2024-12-02:
Lock files are very useful for others trying to replicate your work, identify bugs in new deps and for anyone trying to create a reproducible build.

https://wiki.debian.org/ReproducibleBuilds

Others trying to generate Reproducible can maintain there own lock files but its nice this info is provided by the upstream so that if any bugs are found the reproducible build is using the same set of deps as the upstream.

As this is a library is fair that you don't have a log file but i do build your examples as a test that this lib works on my platform so though i would ask what your thoughts are?

chemicstry, 2024-12-05:
The consensus was that you commit lock file for binaries, but not for libraries, which I followed. But this seems changed now: https://blog.rust-lang.org/2023/08/29/committing-lockfiles.html

I guess we could start commiting the lock file, but that would not make builds reproducibe, because we depend on external library (libcamera) that is not managed by cargo.

fishrockz, 2024-12-05:
Well, for my project (but this also goes for lots of others, yocto, buildstream, some desktop distros) were I lock the version of as much as possible including libcamera then I can use your lock file for libcamera-rs rather than having to have my own lock file rather than one others share. :)

It's not the end of the world if you don't but it is helpful for some people, including me :)