lit-robotics/libcamera-rs #44: Examples do not work with mainline of libcamera and this repo on the pi zero

fishrockz, 2024-12-04:
I tried using the current main of libcamera a43ea7ff70e332ffe6b852a0aaeeb9aa877663cf and main of this repo 1976cea2c62764fbe57e183304f6dbdc368bae5e and get the following error on a pi zero 2 with a PI camera v3

These used to work with older versions, fd84180d7a09eb9f4891f740735b28af68c201af and df2d09f9c03aa5a74352197e076405c1010a6cff

```
-bash-5.2# jpeg_capture test.jpg
[0:10:13.859325599] [399]  INFO Camera camera_manager.cpp:327 libcamera v0.3.2
[0:10:14.029847213] [400]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media2
thread 'main' panicked at libcamera/examples/jpeg_capture.rs:28:54:
called `Result::unwrap()` on an `Err` value: ValueError(InvalidType { expected: 6, found: 8 })
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
-bash-5.2# 
```

```
-bash-5.2# list_cameras                                                                                                                                                                                           
[0:12:52.703752100] [425]  INFO Camera camera_manager.cpp:327 libcamera v0.3.2
[0:12:52.874422152] [426]  INFO RPI vc4.cpp:401 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media2
Camera 0
ID: /base/soc/i2c0mux/i2c@1/imx708@1a
Properties: {
    SystemDevices: String(
        "\rQ\0\0\0",
    ),
thread 'main' panicked at libcamera/src/control.rs:227:62:
called `Result::unwrap()` on an `Err` value: UnknownType(9)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
-bash-5.2# 
```

I'm not sure how to go about debugging these..

chemicstry, 2024-12-05:
This is because libcamera changes propery/control IDs and we can't support every single change. Hence we only support release versions of libcamera (currently up to v0.3.2). See https://github.com/lit-robotics/libcamera-rs/pull/38 for more info

kbingham, 2024-12-05:
Yes, we're currently in an ABI break - so the next release is 0.4.0. I'm hoping to tag it soon - but I'm waiting on either one or now maybe two ABI breaking changes to get merged so all the breakage happens close together.

fishrockz, 2024-12-05:
Ok, no problem at all, it will just use the latest release for now.

I found some very minor bugs in the older versions I was using and wanted to see if they were fixed in main.

I will try again with 0.3.2 and if they are still there I will report them for that release,

There about appart missing matches with colour formats and colour ordering, so nothing to worry about, it might even me me missing understanding.

closed 2024-12-05 by fishrockz