lit-robotics/libcamera-rs #47: Help for the Pi5

fishrockz, 2024-12-20:
Hi have been working on trying to add support for the Pi5

Currently libcamera does not support the Pi5 but the raspberry pi fork does

I have run the libcamera-meta tooling for the main branch of libcamera and i get https://github.com/lit-robotics/libcamera-rs/commit/0d7536a563b762932c037c524b25c21f5195b275

I then ran the tooling for the raspberry pi fork https://github.com/lit-robotics/libcamera-rs/commit/bdd79a62b5bffd698ef7d7d3b14edcf9b1988f01

not i am using the next branch in the pi fork https://github.com/raspberrypi/libcamera/commit/b7ba311b91b53915f20b1d0e188a50658833daa6 and compiling it https://gitlab.com/gorse/heather-linux/-/blob/willsalmon/libcamera-0.4.0/elements/libcamera.grd?ref_type=heads#L137

```bash
meson --buildtype=release --prefix=/usr/ --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dlc-compliance=disabled -Ddocumentation=disabled -Dcam=enabled -Ddocumentation=disabled -Dtest=false -Dandroid=disabled build --cross-file cross_build.txt
```

when i run this on the pi5 

list_cameras works giving a lot of output i can link

```bash
Camera 0
ID: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a
Properties: {
    SystemDevices: SystemDevices(
        [
            20765,
            20766,
            20767,
            20768,
            20769,
            20770,
            20771,
            20739,
            20740,
            20741,
            20742,
        ],
    ),
    ScalerCropMaximum: ScalerCropMaximum(
        Rectangle {
            x: 0,
            y: 0,
            width: 0,
            height: 0,
        },
    ),
    PixelArrayActiveAreas: PixelArrayActiveAreas(
        [
            Rectangle {
                x: 16,
                y: 24,
                width: 4608,
                height: 2592,
            },
        ],
    ),
    PixelArraySize: PixelArraySize(
        Size {
            width: 4608,
            height: 2592,
        },
    ),
    Rotation: Rotation(
        180,
    ),
    Location: CameraExternal,
    ColorFilterArrangement: RGGB,
    UnitCellSize: UnitCellSize(
        Size {
            width: 1400,
            height: 1400,
        },
    ),
    Model: Model(
        "imx708",
    ),
}
Available formats: {
    R8: [
        Size {
            width: 160,
            height: 120,
        },
        Size {
            width: 240,
            height: 160,
        },
        Size {
            width: 320,
            height: 240,
        },
        Size {
            width: 400,
            height: 240,
        },
        Size {
            width: 480,
            height: 320,
        },
        Size {
            width: 640,
            height: 360,
        },
.....
```

There is a LOT of output, i can add if needed

but when i try todo a capture i get

```bash
jpeg_capture test.jpg
[0:00:54.589039688] [390] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:327 [0mlibcamera v0.3.2
[0:00:54.590068947] [391] [1;32m INFO [1;37mRPI [1;34mpisp.cpp:720 [0mlibpisp version 1.0.7
[0:00:54.689249410] [391] [1;32m INFO [1;37mRPI [1;34mpisp.cpp:1179 [0mRegistered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Using camera: imx708
Generated config: [
    StreamConfigurationRef {
        pixel_format: MJPEG,
        size: Size {
            width: 800,
            height: 600,
        },
        stride: 0,
        frame_size: 0,
        buffer_count: 4,
    },
]
[0:00:54.689681873] [390] [1;33m WARN [1;37mV4L2 [1;34mv4l2_pixelformat.cpp:346 [0mUnsupported V4L2 pixel format <INVALID>
[0:00:54.689696095] [390] [1;33m WARN [1;37mFormats [1;34mformats.cpp:1006 [0mUnsupported pixel format 0x00000000
[0:00:54.689693595] [390] [1;31mERROR [1;37mRPI [1;34mpisp.cpp:228 [0mPixel format <INVALID> unsupported
[0:00:54.689704595] [390] [1;35mFATAL [1;37mdefault [1;34mpisp.cpp:229 [0massertion "0" failed in toPiSPImageFormat()
```

I have made a lot of blind faith jumps with the linked patches 

I assume that `Unsupported pixel format 0x00000000` is a pretty bad sign lol, anyone have any clues? i didnt find many hits for using `cargo run --bin generate_from_git` beyond the obvious..

fishrockz, 2025-01-18:
FYI i got the pi working but it needs support for libcamera version 0.4.0

I used this branch of this lib https://github.com/fishrockz/libcamera-rs/tree/willsalmon/0.4.0 to get it working

Please see this basic egui app of a live viewer https://photos.fife.usercontent.google.com/pw/AP1GczMLOz5Lo6UnIhGPCvb8yK1LwCtHT2BnJeoS-2rIcqmwDxZYB2HKekG-=w1502-h2001-s-no?authuser=0

This works with the YUYV feed as the MJPEG feed is not provided by the PI

zhil, 2025-02-09:
@fishrockz thanks a lot for your suggestions and for your fork!

Any chances to see repo of your usage of the YUYV feed instead of MJPEG?

yakir12, 2025-03-25:
@fishrockz , seconding Zhil's request: any chances to see repo of your usage of the YUYV feed instead of MJPEG?

fishrockz, 2025-03-25:
So the formats the view finder can provide come from upstream libcamera so please go open a issue at https://github.com/raspberrypi/libcamera as this is not available for the c++ ether.

One thing that I keep meaning to try is to see what is available in the video stream rather than the preview stream.

I have been working on girderstream recently but will cycle back, I was frustrated by the pi drivers in libcamera as the pi people don't work very closely with libcamera... So hopefully they should have made more progress by now

All that said it does work ok on the pi

[image]

The above is all rust using this lib

fishrockz, 2025-03-25:
Sorry I miss understood your question 

https://gitlab.com/gorse/pi-camera/-/blob/trying/camera-app/crates/pi-camera/src/main.rs?ref_type=heads#L494

It's in the big yellow camera app repo but in branch for now. Hopefully I can cycle back in a bit and mege it in

chemicstry, 2025-05-26:
Thank you for investigating this, libcamera v0.4.0 is now supported as of #60

closed 2025-05-26 by chemicstry