lit-robotics/libcamera-rs #50: Is the CI broken because of using ubuntu:latest and v0.1.0 of libcamera

fishrockz, 2024-12-20:
The ci seems to be broke, it looks like it might be todo with python3.12 not supporting `imp`

by using ubutnu:latest the major versions of software used in the CI can change as ubuntu:latests changes leading to issues like https://github.com/lit-robotics/libcamera-rs/actions/runs/12436691688/job/34725024709?pr=48#step:3:233

I would suggest we fix the version of ubuntu to one with python3.11 or we update from testing libcamera 0.1.0 ?

Which of these would be preferable? does version  0.1.0 represent a golden version for something that is worth testing with?

chemicstry, 2025-05-27:
Gave you write access to this repo.

AFAIK there is no official libcamera group on gitlab, they use selfhosted git and mailing lists for development, which is completely alien to me.

kbingham, 2025-05-27:
https://gitlab.freedesktop.org/camera/ is the official libcamera group. Freedesktop is an organisation which supports open source software which runs the linux desktop ecosystem.

fishrockz, 2025-05-27:
Yer, freedesktop is a great org that is a key pillar of the linux userland eco system.

I do think for the long term maintenance of the repo, freedesktop would be a great place but github and gitlab are fairly similar and github does have better but not perfect discoverability.

I am gona have a think about adding more tests as that would be really helpful for both this repo and for my integration projects that i use to build and test this lib.

If I get to the point were I have some HIL tests ready to go then I would like to revisit this as its much easier for me to get sponsored gitlab runners than github ones.