lit-robotics/libcamera-rs #55: Faster Framerate

UlmerMan, 2025-02-14:
After i ran the video_capture example using:
```
cargo run --example video_capture --release -- ./test.mjpeg
```
i only got a video with 15fps whilst my camera supports 30fps.

kbingham, 2025-02-14:
Is it dark? Are you using a UVC webcam ?

Almost all UVC webcams will reduce frame rate to 15 FPS so that they can double the exposure time to 60 ms in dark environments.

The IPA for raspberry pi may also try to do the same. 

Turn a light on and see if the frame rate goes back up to 30.

UlmerMan, 2025-02-15:
Im using my Laptop built-in webcam. In the spec sheet it says 30fps. Ill try it on i raspberry pi camera 3 later that they as this is the platform my project will run on later.

kbingham, 2025-02-15:
If you were using your Laptop built in webcam, that's definitely the sort of camera I would expect to 'automagically' drop to 15 FPS in low light conditions.

UlmerMan, 2025-02-15:
Now i tried it on a Raspberry Pi 4b with a camera module 3. It recorded with 30fps flawlessly. Sorry if this a dumb question. But how can i record in higher fps. The camera module also supports 50fps according to the datasheet

kbingham, 2025-02-15:
In libcamera, you set the control FrameDurationLimits[min,Max].

The duration is 1/rate. And the minimum frame duration is the fastest rate, and the maximum frame duration is the slowest rate. You can set them the same, for an exact rate, or as we've seen for webcam you can say "between 15 and 30" FPS depending on lighting conditions.

closed 2025-05-20 by UlmerMan