lit-robotics/libcamera-rs #56: support for libcamera 0.4.0?

smroid, 2025-02-15:
I updated my RPi and now I get:

  thread 'main' panicked at /home/cedar/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libcamera-0.3.0/build.rs:49:17:
  Unsupported version of libcamera detected: 0.4.0
  supported versions are: 
        0.0.1
        0.0.2
        0.0.3
        0.0.4
        0.0.5
        0.1.0
        0.2.0
        0.3.0
        0.3.1
        0.3.2

In the meantime I'll look into downgrading my RPi's libcamera back to 0.3.2.

Thanks,
-Steven

funnbot, 2025-02-16:
I don't know if there is a better way to downgrade, but after a good amount of pain I managed it.
```bash
sudo apt remove rpicam-apps-lite libcamera0.4 libcamera0.3 libcamera-ipa libcamera-dev libpisp-common libpisp1

curl https://archive.raspberrypi.org/debian/pool/main/r/rpicam-apps/rpicam-apps-lite_1.5.3-1_arm64.deb -O
curl https://archive.raspberrypi.org/debian/pool/main/libc/libcamera/libcamera0.3_0.3.2+rpt20241119-1_arm64.deb -O
curl https://archive.raspberrypi.org/debian/pool/main/libc/libcamera/libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb -O
curl https://archive.raspberrypi.org/debian/pool/main/libc/libcamera/libcamera-dev_0.3.2+rpt20241119-1_arm64.deb -O
curl https://archive.raspberrypi.org/debian/pool/main/libp/libpisp/libpisp-common_1.0.7-1_all.deb -O
curl https://archive.raspberrypi.org/debian/pool/main/libp/libpisp/libpisp1_1.0.7-1_arm64.deb -O

sudo dpkg -i ./libcamera0.3_0.3.2+rpt20241119-1_arm64.deb ./libcamera-dev_0.3.2+rpt20241119-1_arm64.deb ./libcamera-ipa_0.3.2+rpt20241119-1_arm64.deb ./libpisp1_1.0.7-1_arm64.deb ./libpisp-common_1.0.7-1_all.deb ./rpicam-apps-lite_1.5.3-1_arm64.deb
```

UlmerMan, 2025-02-22:
This didnt work for me. After a ran the commands I still encountered the error

DArtagnant, 2025-03-01:
Same for me. I ran
```bash
sudo apt autoremove
cargo clean
```
And it worked.

UlmerMan, 2025-03-01:
I needed to install libpisp and export the linker path to libcamera

hlima, 2025-03-24:
I have other libraries that depend on libcamera 4.0. Is there any plan to update  this crate to 4.x?

fishrockz, 2025-04-30:
Hi guys, supporting pi's is a bit of a faff as libcamera https://libcamera.org/ upstream and libcamera from the raspberry pi folks https://github.com/raspberrypi/libcamera arnt always very well aligned.. 

I have not been as active with libcamera as i would like to be as i am spending quite a bit of my open source hacking time on my apps or my intergration work but i have recenly updated my 0.4.0 branch to also work with 0.5.0 you can find it here https://github.com/fishrockz/libcamera-rs/tree/willsalmon/0.5.0

and point to it with 

```
libcamera = { git = "https://github.com/fishrockz/libcamera-rs", rev = "5f4db0844756f84905f48d708e7c81cfce4bf474" }
```

I have done a little testing with this one and it generally compiles but the version of 0.5.0 i have been testing with fails on the pi zero with

```
[0:06:54.956484060] [796]  INFO Camera camera.cpp:1205 configuring streams: (0) 400x300-YUYV (1) 4608x2592-RGB888
[0:06:54.958741143] [798]  INFO RPI vc4.cpp:570 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
Allocated 4 buffers
Allocated 1 buffers
[0:06:55.382328383] [798] ERROR V4L2 v4l2_videodevice.cpp:1321 /dev/video14[19:cap]: Unable to request 1 buffers: Cannot allocate memory
[0:06:55.442304320] [798] ERROR RPI pipeline_base.cpp:673 Failed to allocate buffers
thread 'main' panicked at crates/carera-collection/src/main.rs:251:21:
called `Result::unwrap()` on an `Err` value: Os { code: -12, kind: Uncategorized, message: "Unknown error -12" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

But i think thats a issue with the pi-zero + libcamera-0.5.0 rather than with the bindings

Could you guys have a try at this?

@hlima @UlmerMan @DArtagnant  @funnbot @smroid 

I will hopefully find some time to do more detailed testing and get to the bottom of some of these issues at the week end.

fishrockz, 2025-04-30:
If people are able to test could you let me know which version of libcamera your using and how you installed it and also let me know what version pkgconf thinks your using eg

```bash
bash-5.2# pkgconf --modversion libcamera
0.5.0
```

As my branch uses pkgconf to determine which headers to use when linking so it would be really helpful for developing.

Swarkin, 2025-05-05:
@fishrockz hey there, your branch successfully compiled on my raspberry pi with libcamera v0.5.0. Is there anything I need to consider before trying to use your fork of the crate?

fishrockz, 2025-05-17:
Thanks for the feedback @UlmerMan and @Swarkin i have been pretty busy this last few weeks so have not had a chance to get back to this but i can confirm that that branch worked for quite a lot of testing i did with a pi-zero

I have come to this repo after all the hardwork was done by the original authors but my patches are all meant to be pretty universal.

fishrockz, 2025-05-17:
@Swarkin what version of libcamera are you using and what does pkgconf think the version is?

Swarkin, 2025-05-17:
@fishrockz I have `libcamera0.5/stable,now 0.5.0+rpt20250429-1 arm64 [installed,automatic]`, but i abandoned the project for now so i dont need it anymore

fishrockz, 2025-05-18:
@Swarkin thanks for taking the time to get back to us if your not working on this. The more info we can get the easier it is to support more poeple,

kylona, 2025-05-18:
@fishrockz I had similar errors, so maybe I can fill in some details. I may be over-documenting my saga. 

### TLDR 
I was having compile errors in @fishrockz branch that seem to be fixed in @UlmerMan's latest. 

### The saga:

I pulled the branch at https://github.com/fishrockz/libcamera-rs.git

I was seeing a similar compile error as @Swarkin. Running:
```cargo run --example version```

I've been poking around quite a bit so I don't have a good record of how I installed libcamera but I used apt to install several versions of libcamera, then removed them and installed the default version with something like:
```sudo apt install libcamera-dev libcamera-ipa```

Hardware: Raspberry Pi zero 2W. 
OS: Raspbian GNU/Linux 12 (bookworm)
Rustc version: rustc 1.88.0-nightly (17ffbc81a 2025-04-04)
libcamera pkgconf modversion: 0.4.0

I tried with libcamera 0.4.0 first because this was the default version to be installed in raspbian with apt as of a few weeks ago when I started the project. 

Then today I did
```
sudo apt update
sudo apt upgrade libcamera-dev libcamera-ipa
cargo clean
cargo run --example version
```

So my environment looked like this:

Hardware: Raspberry Pi zero 2W. 
OS: Raspbian GNU/Linux 12 (bookworm)
Rustc version: rustc 1.88.0-nightly (17ffbc81a 2025-04-04)
libcamera pkgconf modversion: 0.5.0

But that didn't seem to make a change.

### The happy result

Then I re-read this thread and saw @UlmerMan's branch. I pulled that branch and it compiled successfully first try. 

Big thanks to you @UlmerMan. Do you plan to submit that branch as a PR? (it looks like you have a draft with an older version).

UlmerMan, 2025-05-20:
@kylona nice that it works for you, but I didn't change anything that effects support. It is only @fishrockz 's branch with this [patch](https://github.com/lit-robotics/libcamera-rs/pull/54) applied. I won't make a PR as this was not my work but @fishrockz . I will test it on my own Raspberry Pi Zero when I've got time.

fishrockz, 2025-05-25:
@kylona did you use the default branch of my repo or did you checkout the 0.4.0 or 0.5.0 branch? https://github.com/fishrockz/libcamera-rs/tree/willsalmon/0.5.0

The original maintainers of this lib are very much quite these days

I am happy to work with others to try to support this lib but without the ablity to update the default branch of this repo or to update the crate a fork might be something we should consider

kylona, 2025-05-25:
@fishrockz I may have been on the wrong branch because I just pulled 
```commit 5f4db0844756f84905f48d708e7c81cfce4bf474 (HEAD -> willsalmon/0.5.0, origin/willsalmon/0.5.0)```

And it compiled with no errors. Thanks for your patience. And thanks for keeping this library maintained! 

A fork works if we have to. But if we can catch the attention of someone with access to merge a pull request, that would be best to prevent crates.io from getting cluttered with abandonware. 

@chemicstry Seems to be the original author.  Sometimes the github emails get buried/filtered so a maybe a friendly hello to chemicstry@gmail.com might work.

chemicstry, 2025-05-26:
Fixed in #60, libcamera-rs `v0.4.0` has been released on crates. Thanks everyone for the work.

closed 2025-05-26 by chemicstry