lit-robotics/libcamera-rs #70: Segmentation fault

askolds11, 2025-07-09:
I'm trying to run the `jpeg_capture` example minimally modified on my Pi Zero W, however after the file is written and function is exiting, there's a Segmentation fault.

I am not running it using cargo run, but I cross compile it first - fresh cargo project, copied code from `jpeg_capture.rs`. To run it, I added `drm_fourcc` and changed `PIXEL_FORMAT_MJPEG` to `PixelFormat::new(DrmFourcc::Bgr888 as u32, 0);` as MJPEG does not exist for Pi Camera 2, and removed the MJPEG check.  The file is written correctly (it's raw bytes, but I can manage to get a proper image out of it with some post processing).

For cross compilation I'm using a docker image with armv6 compiler, using readelf I've confirmed that the binary is armv6. (can attach a Dockerfile if needed, very messy as I'm new to docker as well)

I am not very experienced in Rust, much less with C++ stuff being included,  what are my next debugging steps so that I can attach more info here?

chemicstry, 2025-07-10:
Can you try running it with valgrind? I.e. `valgrind ./examplerun Test.jpg`, valgrind should be installable from apt

askolds11, 2025-07-10:
I can't get valgrind to run - the apt version spits out Illegal instruction for any command (e.g. valgrind, valgrind --help, valgrind --version), and I can't figure out how/where to get a different version. I tried compiling it, but on Pi Zero W it spits out `selected architecture lacks an FPU`, I also tried to cross compile it in the docker environment, but I get
```
m_dispatch/dispatch-arm-linux.S: Assembler messages:
m_dispatch/dispatch-arm-linux.S:102: Error: selected processor does not support `movw r1,#47' in ARM mode
m_dispatch/dispatch-arm-linux.S:103: Error: selected processor does not support `movw r2,#0' in ARM mode
m_dispatch/dispatch-arm-linux.S:155: Error: selected processor does not support `movw r4,#:lower16:vgPlain_stats__n_xIndirs_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:156: Error: selected processor does not support `movt r4,#:upper16:vgPlain_stats__n_xIndirs_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:173: Error: selected processor does not support `ubfx r6,r6,#0,#13' in ARM mode
m_dispatch/dispatch-arm-linux.S:176: Error: selected processor does not support `movw r4,#:lower16:vgPlain_tt_fast' in ARM mode
m_dispatch/dispatch-arm-linux.S:177: Error: selected processor does not support `movt r4,#:upper16:vgPlain_tt_fast' in ARM mode
m_dispatch/dispatch-arm-linux.S:204: Error: selected processor does not support `movw r4,#:lower16:vgPlain_stats__n_xIndir_hits1_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:205: Error: selected processor does not support `movt r4,#:upper16:vgPlain_stats__n_xIndir_hits1_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:226: Error: selected processor does not support `movw r4,#:lower16:vgPlain_stats__n_xIndir_hits2_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:227: Error: selected processor does not support `movt r4,#:upper16:vgPlain_stats__n_xIndir_hits2_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:248: Error: selected processor does not support `movw r4,#:lower16:vgPlain_stats__n_xIndir_hits3_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:249: Error: selected processor does not support `movt r4,#:upper16:vgPlain_stats__n_xIndir_hits3_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:258: Error: selected processor does not support `movw r4,#:lower16:vgPlain_stats__n_xIndir_misses_32' in ARM mode
m_dispatch/dispatch-arm-linux.S:259: Error: selected processor does not support `movt r4,#:upper16:vgPlain_stats__n_xIndir_misses_32' in ARM mode
```

askolds11, 2025-07-13:
I didn't manage to get gdb working, shows that auto-load script is missing, once I get some more time I'll try to get it working:
```
pizero@pizero:~/app $ gdb --args examplerun Test.jpg
GNU gdb (Raspbian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from examplerun...
warning: Missing auto-load script at offset 0 in section .debug_gdb_scripts
of file /home/pizero/app/examplerun.
Use `info auto-load python-scripts [REGEXP]' to list them.
```

info `auto-load python-scripts` returned
```
Loaded  Script                                                                 
No      gdb_load_rust_pretty_printers.py    
```

askolds11, 2026-11-07:
Sorry for the massive delay. The project was urgent, so I ended up using python bindings, and I used a different Dockerfile than what I used for this test.
All that to say, I've once again copied the jpeg_example code and did the changes, however I've let Claude generate the Dockerfile and now it works, so I'm 99% sure it was user error. Here are my theories why I had this issue:
1. Version mismatch. My Dockerfile version was 0.5.1, however in the log above it's 0.5.1. I can not be certain of this, as the Dockerfile is from my git history and it might've not been the same one.
2. Wrong Dockerfile (specifically cross compilation) - this time I generated the Dockerfile from scratch using Claude, and while lots of parts differ, the biggest issue might've been my meson cross compilation config for libcamera, at least looking at the Dockerfile in my git history. I had mistakenly considered host_machine as my machine instead of the Raspberry Pi.

As it seems like there is no problem with the library and it was simple user error, I will close the issue.

In case it helps others, I will put my original Dockerfile and associated files below (not the same ones I used back then, but ones I used for compiling with python camera), and Claude generated one too (though I do not recommend it as is, as I asked it to generate it with Jetbrains devcontainer in mind and Claude in it, and I have not gone through it).

Originals:
[pythoncrossbuild.sh](https://github.com/user-attachments/files/34527546/pythoncrossbuild.sh)
[cross-file.txt](https://github.com/user-attachments/files/34527545/cross-file.txt)
[Dockerfile.txt](https://github.com/user-attachments/files/34527551/Dockerfile.txt)

Claude new:
[Dockerfile-claude.txt](https://github.com/user-attachments/files/34527558/Dockerfile-claude.txt)

closed 2026-11-07 by askolds11